tesa-klebeband/RTL8720dn-Deauther #3: having trouble uploading,

michiel1106, 2024-09-07:
I have a bw16 kit, heres the error log:

C:\Users\michi\Downloads\RTL8720dn-Deauther\wifi_cust_tx.cpp: In function 'void wifi_tx_raw_frame(void*, size_t)':
C:\Users\michi\Downloads\RTL8720dn-Deauther\wifi_cust_tx.cpp:12:50: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
   void *frame_control = alloc_mgtxmitframe(ptr + 0xae0);
                                                  ^~~~~
C:\Users\michi\Downloads\RTL8720dn-Deauther\wifi_cust_tx.cpp:15:50: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     update_mgntframe_attrib(ptr, frame_control + 8);
                                                  ^
C:\Users\michi\Downloads\RTL8720dn-Deauther\wifi_cust_tx.cpp:16:50: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     memset((void *)*(uint32_t *)(frame_control + 0x80), 0, 0x68);
                                                  ^~~~
C:\Users\michi\Downloads\RTL8720dn-Deauther\wifi_cust_tx.cpp:17:68: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     uint8_t *frame_data = (uint8_t *)*(uint32_t *)(frame_control + 0x80) + 0x28;
                                                                    ^~~~
C:\Users\michi\Downloads\RTL8720dn-Deauther\wifi_cust_tx.cpp:19:35: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     *(uint32_t *)(frame_control + 0x14) = length;
                                   ^~~~
C:\Users\michi\Downloads\RTL8720dn-Deauther\wifi_cust_tx.cpp:20:35: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     *(uint32_t *)(frame_control + 0x18) = length;
                                   ^~~~
C:\Users\michi\Downloads\RTL8720dn-Deauther\RTL8720dn-Deauther.ino: In function 'void handleRoot(WiFiClient&)':
C:\Users\michi\Downloads\RTL8720dn-Deauther\RTL8720dn-Deauther.ino:130:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (int i = 0; i < scan_results.size(); i++) {
                   ~~^~~~~~~~~~~~~~~~~~~~~
C:\Users\michi\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp: In member function 'virtual uint8_t WiFiClient::connected()':
C:\Users\michi\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp:34:16: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     if ((_sock < 0) || (_sock == 0xFF)) {
          ~~~~~~^~~
C:\Users\michi\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp: In member function 'virtual int WiFiClient::available()':
C:\Users\michi\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp:54:15: warning: comparison is always true due to limited range of data type [-Wtype-limits]
     if (_sock >= 0) {
         ~~~~~~^~~~
C:\Users\michi\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp: In member function 'virtual void WiFiClient::stop()':
C:\Users\michi\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp:126:15: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     if (_sock < 0) {
         ~~~~~~^~~
C:\Users\michi\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp: In member function 'virtual size_t WiFiClient::write(const uint8_t*, size_t)':
C:\Users\michi\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp:139:15: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     if (_sock < 0) {
         ~~~~~~^~~
C:\Users\michi\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp: In member function 'virtual WiFiClient::operator bool()':
C:\Users\michi\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp:157:18: warning: comparison is always true due to limited range of data type [-Wtype-limits]
     return _sock >= 0;
            ~~~~~~^~~~
C:\Users\michi\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp: In member function 'virtual int WiFiClient::connect(const char*, uint16_t)':
C:\Users\michi\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp:176:19: warning: comparison is always false due to limited range of data type [-Wtype-limits]
         if (_sock < 0) {
             ~~~~~~^~~
C:\Users\michi\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp: In member function 'virtual int WiFiClient::connect(IPAddress, uint16_t)':
C:\Users\michi\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp:192:15: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     if (_sock < 0) {
         ~~~~~~^~~
C:\Users\michi\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp: In member function 'virtual int WiFiClient::connectv6(IPv6Address, uint16_t)':
C:\Users\michi\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp:206:15: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     if (_sock < 0) {
         ~~~~~~^~~
Sketch uses 733948 bytes (34%) of program storage space. Maximum is 2097152 bytes.

ilbossdelfaidate, 2024-09-07:
the main question in this case would be what version of sdk did you use to make it work properly?

tesa-klebeband, 2024-09-07:
Hi
To answer your question first, I'm using the latest SDK version available. I cannot see any issues with the log you sent me apart from a few warnings which also occur on my side. Can you send the full log? I believe there should be at least a failure message included after the size reporting. Which dev board exactly are you using? Have you bridged the log UART and upload UART?

ilbossdelfaidate, 2024-09-07:
just to confirm: is it 3.1.6 or 3.1.7?

ilbossdelfaidate, 2024-09-07:
It compiles with some warnings for me but I can't figure out: 
1) it doesn't disconnect any devices from the network.
 2) it only finds 2.4ghz networks

tesa-klebeband, 2024-09-07:
> just to confirm: is it 3.1.6 or 3.1.7?

Probably 3.1.6 if .7 is out just recently, I can test it in a few moments.

tesa-klebeband, 2024-09-07:
No, I am using 3.1.7 and it works as expected. The only difference i can see is that I am using Arch Linux and you seem to be using Windows. Are you able to test if it works if you upload it under Linux? Also, how can you say that uploading doesn't work and then that  it works only for 2.4GhZ?

michiel1106, 2024-09-07:
Okay so im kind of stupid, it wasnt the program, it just wasnt  in uart mode, but idk how to put it in uart mode, any advice? I tried connecting the pins like this

[image]

but yeah, ididnt work

tesa-klebeband, 2024-09-07:
There's an option in the arduino-ide to upload automatically. It's called auto flash mode and under tools and needs to be enabled.

michiel1106, 2024-09-07:
tried that too before but it didnt work, gonna try again

michiel1106, 2024-09-07:
Start Erase Flash 
Erasing...........error: Device not supports Auto Upload Mode

this is just erasing but still

michiel1106, 2024-09-07:
uploading doesnt work either

tesa-klebeband, 2024-09-07:
Try starting the upload and when its in the detecting state, try holding BURN, while doing that pressing and releasing RST and then releasing BURN.

tesa-klebeband, 2024-09-07:
Another thing to try would be using a lower upload baud

michiel1106, 2024-09-07:
doesnt seem to do anything, ill try the lower baud

michiel1106, 2024-09-07:
nope

michiel1106, 2024-09-07:
just to confirm, changing baud is in tools->upload speed right?

michiel1106, 2024-09-07:
[image]

these are my two options

tesa-klebeband, 2024-09-07:
Yes, just try the other upload speed, maybe that works for you. I am using 1500000

michiel1106, 2024-09-07:
I was using the other

michiel1106, 2024-09-07:
both dont work, I tried a bunch of pin configurations n stuff, which do you use to open your port?

michiel1106, 2024-09-07:
omfg im such an idiot, I had the pin layout I posted earlier in reverse, fuckin hell

michiel1106, 2024-09-07:
IT WORKED

michiel1106, 2024-09-07:
I am wondering, the same network has a bunch of different codes, is there a way to attack all of them?

tesa-klebeband, 2024-09-07:
Great! What different codes are you talking about?

michiel1106, 2024-09-07:
when you launch the thingie theres a site, where you select the codes and such

michiel1106, 2024-09-07:
and then deauth

michiel1106, 2024-09-07:
but my wifi has multiple 5ghz networks on the same network

michiel1106, 2024-09-07:
and what are the reason codes?

tesa-klebeband, 2024-09-07:
Deauthing multiple networks is currently not supported, due to a limitation that the wifi channel cannot be switched after it has been initialized once. For the codes, 6 and 23 are working nicely for deauthing, but using all one after another wouldnt make any difference in the deauth

tesa-klebeband, 2024-09-07:
The reason codes just tell the device to be deauthed why it was deauthed

michiel1106, 2024-09-07:
oh, dangit, there is no way to change the channel thing?

tesa-klebeband, 2024-09-07:
I can try checking the libs for the channel limitation but I dunno how long it will take me, gonna look into it in the next days

michiel1106, 2024-09-07:
omg it worked lmao

michiel1106, 2024-09-07:
one channel was good enough apparantly

michiel1106, 2024-09-07:
idk why

tesa-klebeband, 2024-09-07:
Yea, if you want to deauth just one network it is perfectly fine. i thought you wanted to deauth other networks at the same time too, thats what i was talking about

michiel1106, 2024-09-07:
I realize now, there were multiple options bcause the RSSI was different

michiel1106, 2024-09-07:
but there were only 2 channel types, 40 and 10

tesa-klebeband, 2024-09-07:
Ah get it, pretty random that those are the only channels you have in your region

tesa-klebeband, 2024-09-07:
Are you the one, who opened the other issue? If so, I'm just gonna close it

closed 2024-09-07 by tesa-klebeband

From-Noldor, 2024-09-10:
> 天哪我真是个白痴，我之前发的图钉布局是反过来的，见鬼

Hey man, I'm having exactly issue as u did, can you please tell me what's the correct pin layout?

michiel1106, 2024-09-10:
https://github.com/tesa-klebeband/RTL8720dn-Deauther/issues/3#issuecomment-2335688370

heres the comment with the png of the pin layout, make sure to not reverse it or accidentally mirror it. if it doesnt work take out the wites and do it again properly

tesa-klebeband, 2024-09-10:
There are also multiple pinouts for different boards, so check if you are using the correct one