tesa-klebeband/RTL8720dn-Deauther #7: Deauth 2.4Ghz and 5Ghz at the same time

leomacin, 2024-10-09:
Could this firmware deauth 2.4GHz and 5GHz at the same time?
is it true dual-band or it could only send packages to either 2.4GHz or 5GHz?

thank you

tesa-klebeband, 2024-10-09:
Yes, if you use wext_set_channel(WLAN0_NAME, channel); you can change the channel before transmitting the packets to 2.4 or 5GHz. I may make it possible to deauth multiple networks at the same time in this repo.

leomacin, 2024-10-09:
based on you code, when running deauth, you will lose connection with RTL8720dn-Deauther since the channel will change according to target's channel. is it true?

tesa-klebeband, 2024-10-10:
Yes, but if you would focus on a single channel, it would be possible to close the WiFi and then reopen it after the channel has been set. This wouldn't work however if I chose to implement the capability to deauth multiple channels.

closed 2024-10-22 by tesa-klebeband