tesa-klebeband/RTL8720dn-Deauther #8: Can't get the device to scan 5Ghz networks

Cervidae71, 2024-10-10:
Hi, i'm using the default code on a RTL8720dn board. The original firmware was erased and uploading the sketch worked. Once running, i connect to the accesspoint, login but it seems only to detect 2.4ghz networks. I have tried setting up 1 and 2 5ghz accesspoints but they remain undetected as well others in the neighbourhood. Could it be the board? or is it a setting in the code to enable scanning 5ghz?

tesa-klebeband, 2024-10-10:
Hi, are you sure you selected the right board when uploading? I know there is a similar sounding board without 5G support. Please check. If it doesn't work this way, it might be a problem with the board you are using.

Cervidae71, 2024-10-10:
Hi tnx for responding so fast. i have the ameba ai thinker 8720dn board installed and selected. It might be the board then. try to get another one.

tesa-klebeband, 2024-10-10:
Maybe you could try posting a screenshot of the arduino-ide's settings and I could look over it, cause just the 5Ghz transmitter and receiver being defect and everything else working would be really weird.

ilbossdelfaidate, 2024-10-10:
purple module don't work hardware issues , black mordule work

Cervidae71, 2024-10-12:
The problem is solved. It seems as mentioned above, that the purple boards don't work. I got myself new ones which are green and worked instantly.

closed 2024-10-12 by Cervidae71

hoangminhhp234, 2024-10-24:
> purple module don't work hardware issues (wrong pinout on pcb , wrong connection internal to bw16 module and , diplexer soldered upsidedown), black mordule work

Do you have any way to fix this hardware issue?

ilbossdelfaidate, 2024-10-24:
if you have a microscope we are talking about 0.05mm pieces and traces, a good soldering iron and you can uncover it without destroying anything you should make many bridges with 0.01mm wires invert the diplexer and cut various tracks with the risk of destroying everything, considering the cost, it is better to buy a black module (already tested and it works) or a green module (they say it works but I have not tried it) I would like to point out that I have good equipment for SMD and BGA and on the 4th bridge the tracks were destroyed