tesa-klebeband/RTL8720dn-Deauther #9: Tutorial

pluralcell, 2024-10-12:
Im new to this microcontroller stuff, so if anyone can do a step by step video on how to install this code to my rtl8720dn board. Thank you

tesa-klebeband, 2024-10-12:
Not that good on making videos but I can try explaining it more detailed 😅
1. Download and install the [Arduino IDE](https://www.arduino.cc/en/software)
2. Add the ameba board manager URL to the IDE by opening the file -> preferences section and pasting this into the field on the bottom: [https://github.com/ambiot/ambd_arduino/raw/master/Arduino_package/package_realtek_amebad_index.json](https://github.com/ambiot/ambd_arduino/raw/master/Arduino_package/package_realtek_amebad_index.json)
3. Exit the preferences window and go to the board tab on the left side of the ide, then search for the board and install the most frequent version
4. If you have git installed, clone this repo using: git clone https://github.com/tesa-klebeband/RTL8720dn-Deauther.git. in case you dont have git click download zip and extract the zip locally.
5. Open the main directory with all the code files in the Arduino IDE, select your board at the top and then change it to auto flash mode.
6. Depending on which RTL8720dn development board you have, you need to get your hands on it's pinout. These are the most common pinouts I found online: 
[image]
[image]
7. Connect Log Rx and Log Tx which you can find in the mentioned above pinout to Serial Rx and Serial Rx (same as LP Rx / Tx) on a breadboard and put your  devboard on top of it
8. Connect your rtl8720dn to your pc with a usb cable, select the port if not automatically detected in the ide and hit upload.
9. If everything uploaded successfully, you can refer to the README.md file on how to use it

Hope this helps, have a nice day

pluralcell, 2024-10-12:
Thank you, i'll try my best.

closed 2024-10-12 by pluralcell