tesa-klebeband/RTL8720dn-Deauther #10: plase help to added display

sulintang, 2024-10-21:
You have great project, 

but can you help added oled display?

thank you.

tesa-klebeband, 2024-10-22:
Hi
I'm planning on doing that on a way bigger deauther project in the future, but for now, I might be able to help you. If you could find a library for the OLED you are using, I'll help you further integrate it into the code.
Have a nice day

sulintang, 2024-10-22:
hi,

thank you so much for your replay, i appreciate it.  by the way i have 1.3" SH1106

Have a nice day

tesa-klebeband, 2024-10-23:
Searching for a bit led me to this lib: [Adafruit-GFX-Library](https://github.com/adafruit/Adafruit-GFX-Library)
I didn't find an example on your specific display, but I found the display mentioned with this lib so there's probably an example too if you need one. The lib should work with the rtl8720dn. I don't have this display, so I cannot try it for myself. You could maybe just start trying to experiment with it in a separate project and then, depending on what you want to display on it, integrate it into the code.

sulintang, 2024-10-25:
thank you so much, i'll try

kuku-keke, 2024-11-25:
你们好
我来自于中国的甘肃省，非常感谢您提供的项目文件，但是对于十几岁的我来说上传程序有点困难，上传一直显示错误代码，希望您能出一个教程，我在浏览器上在了好久还是找不到相关内容，我在中国一直支持你们。

kuku-keke, 2024-11-25:
你们好，我们可以交个朋友吗

tesa-klebeband, 2024-11-25:
Hello!
Thank you for pointing this out. I have already given a quick tutorial on doing that in [this issue](https://github.com/tesa-klebeband/RTL8720dn-Deauther/issues/9). I hope, you like my project. If you have any more questions, feel free to ask (I have Discord too). Have a nice day!

kuku-keke, 2024-11-30:
你好,我开了加速器添加开发板管理器地址还是404是怎么回事，还有就是项目需要全部添加吗？还是只需要添加一部分