tesa-klebeband/RTL8720dn-Deauther #11: Web

samukas81, 2024-10-22:
Hallo, können Sie die Firmware-Installation wie bei Ihrem ESP32 über das Internet bereitstellen?  Danke.

tesa-klebeband, 2024-10-22:
Hi,
Hatte mir das auch schonmal überlegt, aber es gibt noch keine Implementation vom webflasher für den Realtek MCU. Muss mal schauen was sich da machen lässt.

samukas81, 2024-10-22:
Ok, danke für die Antwort und Glückwunsch zur Arbeit!

samukas81, 2024-10-22:
Ich erhalte diese Fehlermeldung. Können Sie mir helfen?
[image]

tesa-klebeband, 2024-10-22:
Sieht so aus als ob Sie wifi_cust_tx.cpp und den gleichnamigen Header nicht in ihrem Projekt haben. Am besten Sie clonen das projekt über Git oder laden die zip runter.

tesa-klebeband, 2024-10-22:
Super! Einen schönen Tag noch

closed 2024-10-22 by tesa-klebeband