tesa-klebeband/RTL8720dn-Deauther #12: Adding "deauth all networks" feature

YoavLevi, 2024-10-22:
Hello and thank you Tesa for this project.
I've been playing with it trying to make it deauth all networks around however for some reason it does not deauth all the devices at each network as it's done via the single deauth button of the HTML server.
I was trying to do something like this:
[image]
However, I'm not sure why it's not working well.
Do you have any idea how to implement this feature more successfully?

tesa-klebeband, 2024-10-23:
Hi,
I believe it doesn't work because sending a single frame (a single call to the tx function) might not be enough. I have mentioned in another issue, that reopening the WiFi, as startDeauth does it, is really slow. You can use wext_set_channel to set the WiFi channel before transmitting a few frames, then repeat it with another networks channel...

YoavLevi, 2024-10-23:
Thank you for replying to me. I want to try what you've suggested however I couldn't find out what is the interface name of the WiFi in the BW16 board, as needed by the declaration of the function you've mentioned:

int wext_set_channel(const char *ifname, __u8 ch);

Do you know what value should be substituted to the  *ifname variable?

tesa-klebeband, 2024-10-23:
The name for the WiFi interface should be WLAN0_NAME

YoavLevi, 2024-10-26:
Thanks, it worked!
After only changing the channel for each network the deauth packets are sent more rapidly and successfully act on all networks.

closed 2024-10-26 by tesa-klebeband