tesa-klebeband/RTL8720dn-Deauther #18: ip: 192.168.1.1 cannot access RTL7820dn

Nam-Nobi, 2024-11-21:
I can't access 192.168.1.1 Please show me how to access your RTL8720dn!

tesa-klebeband, 2024-11-22:
You are connected to the rtl8720dn right? Can you verify it not working with another device, a phone for example?

gpvn07, 2024-11-22:
I have tried with all the devices I have: PC, laptop, smartphone all have
the same problem and cannot connect

tesa-klebeband, 2024-11-22:
Can you check which IP address the gateway has? Try to type in this ip

gpvn07, 2024-11-22:
[image]
I can't check the port and when I run bw16 the serial monitor reported

gpvn07, 2024-11-22:
Which ip do I need to try?

gpvn07, 2024-11-22:
I can't check the port and when I run bw16 the serial monitor reported
[image]

tesa-klebeband, 2024-11-23:
Please check which IP is assigned to you and the gateway ip, then post it here.

Khoa5555, 2024-12-10:
I had the same problem while using it. I unplugged it and plugged it back in but it didn't break the wifi. In the end, I had to reload the code. Can you fix it?

Khoa5555, 2024-12-10:
I was also using it. I unplugged it and plugged it back in but it didn't deauther the wifi. Finally, I had to reload the code. Can you fix it?

tesa-klebeband, 2024-12-14:
I have no idea why it isn't working for you. I can only suggest you update the Arduino IDE and try a different device.

closed 2024-12-14 by tesa-klebeband

newer2p, 2025-07-29:
Any updates? > I was also using it. I unplugged it and plugged it back in but it didn't deauther the wifi. Finally, I had to reload the code. Can you fix it?