tesa-klebeband/RTL8720dn-Deauther #20: Help my oled code

warwick320, 2024-11-26:
#undef max
#undef min
#include "vector"
#include "wifi_conf.h"
#include "map"
#include "wifi_cust_tx.h"
#include "wifi_util.h"
#include "wifi_structures.h"
#include "debug.h"
#include "WiFi.h"
#include "WiFiServer.h"
#include "WiFiClient.h"
#include <Wire.h>              
#include <Adafruit_GFX.h>      
#include <Adafruit_SSD1306.h>   
#define SCREEN_WIDTH 128
#define SCREEN_HEIGHT 64
#define OLED_RESET -1
#define BTN_DOWN PA12
#define BTN_UP PA27
#define BTN_OK PA13
#define BTN_MENU PB2
bool deatuhWifiPushBackOnce = false;
String SSIDCh;
bool deatuher_rr = false;
int deauth_reason2 =0;
int deauth_reason3 = 0;
// LEDs:
//  Red: System usable, Web server active etc.
//  Green: Web Server communication happening
//  Blue: Deauth-Frame being sent
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);
const unsigned char menu_attack [] PROGMEM = {
	0x3f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xfc, 
	0x40, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x02, 
	0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
	0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
	0x80, 0x30, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
	0x80, 0x30, 0x00, 0xe0, 0x00, 0x00, 0x00, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
	0x80, 0x30, 0x01, 0x11, 0x04, 0x00, 0x00, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
	0x80, 0x30, 0x01, 0x13, 0xcf, 0x1c, 0x38, 0x88, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
	0x80, 0x30, 0x01, 0x11, 0x04, 0x22, 0x44, 0x90, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
	0x80, 0x30, 0x01, 0x11, 0x04, 0x02, 0x40, 0xa0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
	0x80, 0x30, 0x01, 0xf1, 0x04, 0x1e, 0x40, 0xc0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
	0x80, 0x30, 0x01, 0x11, 0x04, 0x22, 0x40, 0xa0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
	0x80, 0x30, 0x01, 0x11, 0x04, 0x22, 0x44, 0x90, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
	0x80, 0x00, 0x01, 0x10, 0xc3, 0x1e, 0x38, 0x88, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
	0x80, 0x30, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
	0x80, 0x30, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
	0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
	0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 
	0x40, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x02, 
	0x3f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xfc, 
	0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x00, 0x00, 0x00, 0xe0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x00, 0x00, 0x01, 0x10, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x07, 0xff, 0x01, 0x01, 0xc3, 0x8f, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x00, 0x00, 0x01, 0x02, 0x24, 0x48, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x03, 0xfe, 0x00, 0xe2, 0x00, 0x48, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x00, 0x00, 0x00, 0x12, 0x03, 0xc8, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x01, 0xfc, 0x00, 0x12, 0x04, 0x48, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x00, 0x00, 0x01, 0x12, 0x24, 0x48, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x00, 0x70, 0x00, 0xe1, 0xc3, 0xc8, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
	0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
[... 801 lines cut ...]
  scanNetworks();
}

void DrawAttack(){
  display.clearDisplay();
  display.drawBitmap(0,0,menuAttackon,92,64,WHITE);
  display.setCursor(77,30);
  display.setTextSize(1);
  display.setTextColor(SSD1306_WHITE);
  if (SelectedSSID.length() == 0){
    display.print("#HIDDEN#");
  }
  else if (SelectedSSID.length() > 6){
    SelectedSSID = SelectedSSID.substring(0,6) + "..";
    display.print(SelectedSSID);
  }
  else display.print(SelectedSSID);
  display.display();
  if(!deatuhWifiPushBackOnce){
    deauth_wifis.push_back(ScrollIndex);
  }
  deatuhWifiPushBackOnce = true;
  delay(10);
}
This is my SSD1306 OLED code, but it doesn't work. What is the problem?

tesa-klebeband, 2024-11-27:
Could you please state more information about what isn't working? Maybe try some debug statements in the serial to figure out if the flow is going to your liking. I don't really have experience with the OLED lib. Try asking chatgpt or copilot if you have it, as I am not responsible for fixing your code. Don't get me wrong, I'd always like to help, but saying 'fix it' is inappropriate in my opinion.

warwick320, 2024-11-27:
I'm really sorry if I seemed rude. I'm still a student, and my English is not good yet. I truly didn't mean to offend you.

warwick320, 2024-11-27:
The problem I am having with my code is that after deauth_wifis.push_back(ScrollIndex); the code stops and does not work.

warwick320, 2024-11-27:
I apologize in advance if I said anything rude again.

tesa-klebeband, 2024-11-27:
No problem, still a student too :)
Just been getting really tired of getting some weird issues, but I just saw that you have forked my repo which I'm very happy about. I'll look over your code in a couple hours. Have a nice day until then.

tesa-klebeband, 2024-11-27:
Hi! Are you able to tell at which line exactly the code stops working. Also, from what I have had issues in the past, if you do serial printouts, you should initialize the serial in advance. Maybe it helps to clear the deauth wifi vector too before you process the post data into it.

warwick320, 2024-11-28:
The line where the error occurs is line 898. The code stops at this line. Thanks for your reply

China-xiongwei, 2024-12-12:
在Ardoion中上传错误谁能帮帮我 我是新手
[image]

tesa-klebeband, 2024-12-14:
Did you manage to fix it?

closed 2025-01-31 by tesa-klebeband