tesa-klebeband/RTL8720dn-Deauther #22: summary of errors

namkkakkaka, 2024-12-01:
1. When scanning for wifi, you will get an error not found! at 192.168.1.1/rescan 2. When deauth is performed, all nearby wifi is affected, not just the wifi in the selected list 3. Sometimes refreshing the page will also result in a not found error! at website 192.168.1.1

tesa-klebeband, 2024-12-01:
Hello, I already noticed the not found errors and will fix them in the next few days. However, thanks for pointing out the error where all wifis are affected. I myself didn't encounter it, but it might be due to a different browser etc. Could you please post your exact setup, so I can test it myself to fix it.

namkkakkaka, 2024-12-01:
@tesa-klebeband when I select the wifi network in the scanned wifi list and enter reason 2 in the reason list and press start deauth then the wifi networks that I did not select are also affected such as reduced speed or cannot connect

tesa-klebeband, 2024-12-01:
Okay, could you post which browser you are using. Might be an issue with creating the post request and processing it.

namkkakkaka, 2024-12-01:
@tesa-klebeband I use chrome browser on android phone

tesa-klebeband, 2024-12-14:
Reduced speed is normal, as the code is currently sending many frames. You can change the number and time between frames in the loop() function, and check if that fixes it for you.

namkkakkaka, 2024-12-14:
@tesa-klebeband So sending too many frames will affect networks that are not in the selected list.

tesa-klebeband, 2024-12-14:
Yeah kinda, it slows down the other networks speed too, but it's weird that is doesn't let you connect to them either. Also, only the networks on the channel of the to be deauthed network should be affected.

namkkakkaka, 2024-12-14:
@tesa-klebeband yes the network not selected sometimes can not connect or will show no internet

tesa-klebeband, 2024-12-15:
Try changing the speed the frames are sent in
```cpp
delay(5);
```
to a higher value like 10.

droid4gamers, 2024-12-15:
@tesa-klebeband how can I hide the SSID? Thanks!

tesa-klebeband, 2024-12-16:
Hi, I believe you can hide it in an argument to the WiFi.apbegin() function. I'm not sure which one, it should be written in the docs, or in the function definition. If you need any more help, feel free to ask.

droid4gamers, 2024-12-16:
Hello @tesa-klebeband thank you for replying. By any chance do you have a source website where I can see what command to put on how to hide my SSID? Sorry if you find my question lame. I don't have any background on coding or programming. Just searching on Google and YouTube. I'm really noob on this. Thanks again for your response. Btw I appreciate you sharing this.

droid4gamers, 2024-12-19:
> Hi, I believe you can hide it in an argument to the WiFi.apbegin() function. I'm not sure which one, it should be written in the docs, or in the function definition. If you need any more help, feel free to ask.

Thank you @tesa-klebeband. It's all good now.

closed 2025-01-02 by tesa-klebeband

uacode, 2025-01-05:
@tesa-klebeband Hello in html you should send the network as array (for multiple usage)
And correctly process it in a loop (how I don't know exactly, I'm not writing for Arduino)

`response += "<td><input type='checkbox' name='network[]' value='" + String(i) + "'></td>";`