tesa-klebeband/RTL8720dn-Deauther #25: whenever I make multiple selections, regardless of which WiFi I choose, it will be attacked.

Faith-dotw, 2024-12-21:
Hello, as a student, my English isn't very good, so please excuse me if there are some inappropriate words. The issue I'm encountering is that whenever I make multiple selections, regardless of which WiFi I choose, it will be attacked. I saw in other Q&A sessions that trying to change the frame transmission rate might help, but I've done that and it still hasn't made a difference.

tesa-klebeband, 2024-12-21:
Hi! Are you able to modify the code, so it prints out the selections to the serial after the attack post request? I'm currently really busy, so it would be great if you could try debugging it a bit for yourself if that's okay.
Have a nice day

Faith-dotw, 2024-12-22:
I tried to change the code to make it output the WiFi list on the serial port but it caused the crash of the whole program, I hope you can find the reason and modify it after you are not busy, thank you very much

tesa-klebeband, 2025-01-31:
Sorry for my late reply. I tried replicating your issue, but was unable to do so. Maybe you can check other already closed issues or use a different board.

closed 2025-01-31 by tesa-klebeband