tesa-klebeband/RTL8720dn-Deauther #26: How to change default ip address

ycancook, 2025-01-01:
I'm a newbie, I've been learning a lot about the RTL8720 kit lately but can't find a way to use the default ip change code 192.168.1.1. Can anyone help me?

tesa-klebeband, 2025-01-01:
I believe you can use WiFi.config(...) with appropriate arguments as declared in the function definition: 
```cpp
void config(IPAddress local_ip, IPAddress dns_server, IPAddress gateway, IPAddress subnet);
```
Hope this works. Let me know if you have any more questions.

ycancook, 2025-01-02:
> I believe you can use WiFi.config(...) with appropriate arguments as declared in the function definition:
> 
> ```c++
> void config(IPAddress local_ip, IPAddress dns_server, IPAddress gateway, IPAddress subnet);
> ```
> 
> Hope this works. Let me know if you have any more questions.

I succeeded, I posted it for anyone who needs it. Thank you very much

`IPAddress local_ip(192, 168, 5, 1);
IPAddress gateway(192, 168, 5, 1);
IPAddress subnet(255, 255, 255, 0);
IPAddress dns_server(8, 8, 8, 8);
char *ssid = "wifi name";
char *pass = "wifi pass";`

`void setup() {
  Serial.begin(115200);
  WiFi.config(local_ip, dns_server, gateway, subnet);
  WiFi.begin(ssid, pass);
}`

tesa-klebeband, 2025-01-02:
Great! Have a nice day

closed 2025-01-02 by tesa-klebeband