tesa-klebeband/RTL8720dn-Deauther #27: Doesn't deauth any of my 5GHZ networks?

arasakatyler, 2025-01-11:
Hello,

I uploaded the program to my board, and I can see the WiFi network "deauther" and I join it and works just fine

I open to 192.168.1.1 in my browser and see the list of networks including the 5GHz.

I selected all 5ghz networks and put reason "2" and attacked.

I selected all my 5Ghz networks (including old  test dlink router and connected my phone to each AP but I don't see any deauth and no loss of connection on any of the 5Ghz networks?

The board is blinking red blue red blue looks like to be sending deauth frames. 

I know some router uses WPA3 and etc, but I even used my old DIR-60l dlink router which is very old and does have 5Ghz and no WPA3 but still no deauth when my phones connected to the AP and or any of my other 5GHz wifi networks.

Any ideas what's going on?

thanks

tesa-klebeband, 2025-01-11:
Hmm that's weird. If you're using Linux, you can try installing the aircrack-ng utilities and putting your WiFi card, if you happen to use WiFi on your pc, into monitor mode. Then, with wireshark you can try checking your WiFi card for deauth frames. You can also try using different reason codes as discussed in other issues. Is only 5G the issue or 2.4G as well? If so, then there might be an issue with your board (see other issues). If you happen to have another board please try it. You may also try adding debug statements especially on the sections of the code where the WiFi network selections are parsed from html and selected.

arasakatyler, 2025-01-12:
So testing again.. It it deauths 2.4 no problems works right away with any reason code. 

For 5GHz doesn't matter what reason code or what 5Ghz I try it on nothing happens. I'm wondering if these two boards I got off aliexpress that are labled as BW16 and has the correct markings on the Wifi Chip is actually fake and doesn't support 5Ghz. 

On the chip it does read "B&T BW16 WIFI RTL8720DN ARM KM4 FN2 2.4G/5.8G/BT" but yeah no luck, only on 2.4ghz it seems to work. 

I have as well selected only one network and tried multiple but same result for 5ghz. 

I purchased my board from - https://www.aliexpress.com/item/1005001320366011.html?channel=twinner  and does show 80+ sold but no reviews about 5ghz so can't verify. 

darn darn darn. Do you know of any other places to buy a legit board?

I can try to look at the airmon-ng packets etc later but I have a feelings it's the board maybe.

tesa-klebeband, 2025-01-12:
I got mine from Amazon but from a Chinese seller as well. However, mine looks identical to yours. In another [issue](https://github.com/tesa-klebeband/RTL8720dn-Deauther/issues/8) it was ruled out that purple boards don't seem to work because of a hardware connection issue, maybe that is the case with yours as well.

arasakatyler, 2025-01-12:
ah ok, Mine is the black version as well. could be due to country code or something else in the source code.. it's so odd. seems like everything is right and we do see the red/blue blinking when I select 5ghz. hmm. here is a picture of my boards. 
[image]

arasakatyler, 2025-01-12:
So I have 2 ALFA wifi cards. 

I attached 1 to my Kali linux machine (monitor mode with wireshark) and the 2nd Alfa connected to my host windows machine to connect to dev board web interface.

If I plug in the RTL dev board, connect to RTL deauth wifi SSID  on my windows machine and select 2.4Ghz network and attack, we can see on my kali machine (monitor mode) that we do see deauth packets when selecting 2.4ghz networks.(first picture below)
[image]

Now if we restart the dev board, and this time select 5ghz networks and hit attack, we don't see any deauths at all nothing. but the board is still blinking red/blue (second picture)
[image]

hmm.

arasakatyler, 2025-01-12:
I was able to get it to work! I actually randomly looked on youtube and found a video here - https://www.youtube.com/watch?v=odyqsiVz3Dc using your firmware. He also had some issues he said trying to get 5Ghz to work,

It looks like he had to play around in source code with 
                                                                                          uint16_t deauth_reason = 6;
                                                                                           #define FRAMES_PER_DEAUTH 15
He had his at deauth_reason = 2 (normal)   and FRAMES_PER-DEAUTH to 10 and it worked for him

Personally I had to set mine like above    6 & 15. (I tried 10 but didn't work but 15 did) so I guess we just have to play around with these numbers for our device or something like that. either way we got it working with 6 & 15 changed numbers! So 5GHz does indeed work!

I thought it was my board lol

So if happens to anyone else I guess just play around with the numbers. 

Anyways, Thank you! we close it now

arasakatyler, 2025-01-12:
Deleted last comment.. It was actually the 2.4ghz my phone was connected too.. seems to still not work.

tesa-klebeband, 2025-01-12:
I just saw your deleted comment, weird that even after changing the frames per deauth it doesn't work. The deauth reason variable is not really important to change as it gets overwritten by the web interface. No idea what it could be, maybe try looking through the closed issues as well. I think the board might not be the problem as you said that 5G networks are appearing on your list, so the 5G radio does indeed work.

arasakatyler, 2025-01-12:
True, yes indeed the 5Ghz networks do show up in the list, so I guess that does mean the radio is working. 

Very odd. I did go through the other closed issues but nothing seemed to work or change. 

Just to verify, This wouldn't work on WiFi 5ghz networks with WPA3 active right? only ones using WPA2?

I am testing this on an old dlink router with 5ghz, as all my other routers have WPA3.   Maybe I need to try another 5ghz router like Tenda or something.

arasakatyler, 2025-01-12:
You can go ahead and close this case. I set my deauth frames = 30. It's working. I forgot the 2.4ghz on my phone and only made sure 5ghz was connected, and we do indeed see it disconnecting reconnecting disconnecting haha. No idea but looks like we are good now at 30. Thank you!

tesa-klebeband, 2025-01-12:
Great that you were able to figure it out. I'll may sure to update the code in the near future so things like that hopefully won't happen anymore.

closed 2025-01-12 by tesa-klebeband