tesa-klebeband/RTL8720dn-Deauther #28: the download is successful but it does not create an access point

xkrayxbr, 2025-01-13:
init_thread(58), Available heap 0x26060ble_common_init 0 

salve_add_server 

GAP_INIT_STATE_STACK_READY 

app_handle_gap_msg: subtype 1

salve_handle_dev_state_evt: init state 1, adv state 0, cause 0x0

[BLE peripheral] GAP stack ready

app_handle_gap_msg: subtype 1

salve_handle_dev_state_evt: init state 1, adv state 1, cause 0x0app_handle_gap_msg: subtype 1

salve_handle_dev_state_evt: init state 1, adv state 2, cause 0x0
[image]

tesa-klebeband, 2025-01-13:
Looks like the error is not successful after all, see **error: Enter UART Download Mode**. Try to enable auto flash mode under **Tools** and then upload again. Alternatively, you can reset your board while holding the BOOT button and thus enter flash mode manually after which you can upload again. However the first method is easier and doesn't require doing this every upload.

tesa-klebeband, 2025-01-13:
Can you give me the serial logs? Maybe try turning off your WiFi and turning it back on, it can take a while for a phone or pc to register the new network.

xkrayxbr, 2025-01-13:
[image]
this is everything that always appears

xkrayxbr, 2025-01-14:
and the first thing I do after downloading

xkrayxbr, 2025-01-14:
Dude, I don't know what I did, but the firmware started working normally again, thank you for your support and quick response.

tesa-klebeband, 2025-01-14:
Alright, if everythings working now, I'm going to close the issue.

closed 2025-01-14 by tesa-klebeband