tesa-klebeband/RTL8720dn-Deauther #30: DEBUG_SER_PRINT error

supermaaar, 2025-01-16:
C:\Users\PC\Documents\RTL8720dn-Deauther\RTL8720dn-Deauther.ino:11:27: fatal error: DEBUG_SER_PRINT: No such file or directory
 #include "DEBUG_SER_PRINT"
                           ^
compilation terminated.
exit status 1

Compilation error: DEBUG_SER_PRINT: No such file or directory

supermaaar, 2025-01-16:
I didnt touch anything, but everytime I verify it shows the error.

C:\Users\PC\Downloads\RTL8720dn-Deauther-master\RTL8720dn-Deauther-master\RTL8720dn-Deauther\RTL8720dn-Deauther.ino: In function 'int scanNetworks()':
C:\Users\PC\Downloads\RTL8720dn-Deauther-master\RTL8720dn-Deauther-master\RTL8720dn-Deauther\RTL8720dn-Deauther.ino:56:51: error: 'DEBUG_SER_PRINT' was not declared in this scope
   DEBUG_SER_PRINT("Scanning WiFi networks (5s)...");
                                                   ^
C:\Users\PC\Downloads\RTL8720dn-Deauther-master\RTL8720dn-Deauther-master\RTL8720dn-Deauther\RTL8720dn-Deauther.ino: In function 'void setup()':
C:\Users\PC\Downloads\RTL8720dn-Deauther-master\RTL8720dn-Deauther-master\RTL8720dn-Deauther\RTL8720dn-Deauther.ino:279:18: error: 'DEBUG_SER_INIT' was not declared in this scope
   DEBUG_SER_INIT();
                  ^
C:\Users\PC\Downloads\RTL8720dn-Deauther-master\RTL8720dn-Deauther-master\RTL8720dn-Deauther\RTL8720dn-Deauther.ino: In function 'void loop()':
C:\Users\PC\Downloads\RTL8720dn-Deauther-master\RTL8720dn-Deauther-master\RTL8720dn-Deauther\RTL8720dn-Deauther.ino:311:28: error: 'DEBUG_SER_PRINT' was not declared in this scope
     DEBUG_SER_PRINT(request);
                            ^

exit status 1

Compilation error: 'DEBUG_SER_PRINT' was not declared in this scope

tesa-klebeband, 2025-01-16:
You need to download the whole project as zip file and then extract everything to one directory. It should work then.

supermaaar, 2025-01-16:
Did reinstall Arduino.
Download the zip file>extract the file> but still the same error. 
debug.h was there all files are there :(

c:\Users\PC\Documents\Arduino\libraries\RTL8720dn-Deauther-master\wifi_cust_tx.cpp: In function 'void wifi_tx_raw_frame(void*, size_t)':
c:\Users\PC\Documents\Arduino\libraries\RTL8720dn-Deauther-master\wifi_cust_tx.cpp:12:50: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
   void *frame_control = alloc_mgtxmitframe(ptr + 0xae0);
                                                  ^~~~~
c:\Users\PC\Documents\Arduino\libraries\RTL8720dn-Deauther-master\wifi_cust_tx.cpp:15:50: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     update_mgntframe_attrib(ptr, frame_control + 8);
                                                  ^
c:\Users\PC\Documents\Arduino\libraries\RTL8720dn-Deauther-master\wifi_cust_tx.cpp:16:50: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     memset((void *)*(uint32_t *)(frame_control + 0x80), 0, 0x68);
                                                  ^~~~
c:\Users\PC\Documents\Arduino\libraries\RTL8720dn-Deauther-master\wifi_cust_tx.cpp:17:68: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     uint8_t *frame_data = (uint8_t *)*(uint32_t *)(frame_control + 0x80) + 0x28;
                                                                    ^~~~
c:\Users\PC\Documents\Arduino\libraries\RTL8720dn-Deauther-master\wifi_cust_tx.cpp:19:35: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     *(uint32_t *)(frame_control + 0x14) = length;
                                   ^~~~
c:\Users\PC\Documents\Arduino\libraries\RTL8720dn-Deauther-master\wifi_cust_tx.cpp:20:35: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     *(uint32_t *)(frame_control + 0x18) = length;
                                   ^~~~
C:\Users\PC\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp: In member function 'virtual uint8_t WiFiClient::connected()':
C:\Users\PC\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp:34:16: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     if ((_sock < 0) || (_sock == 0xFF)) {
          ~~~~~~^~~
C:\Users\PC\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp: In member function 'virtual int WiFiClient::available()':
C:\Users\PC\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp:54:15: warning: comparison is always true due to limited range of data type [-Wtype-limits]
     if (_sock >= 0) {
         ~~~~~~^~~~
C:\Users\PC\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp: In member function 'virtual void WiFiClient::stop()':
C:\Users\PC\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp:126:15: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     if (_sock < 0) {
         ~~~~~~^~~
C:\Users\PC\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp: In member function 'virtual size_t WiFiClient::write(const uint8_t*, size_t)':
C:\Users\PC\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp:139:15: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     if (_sock < 0) {
         ~~~~~~^~~
C:\Users\PC\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp: In member function 'virtual WiFiClient::operator bool()':
C:\Users\PC\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp:157:18: warning: comparison is always true due to limited range of data type [-Wtype-limits]
     return _sock >= 0;
            ~~~~~~^~~~
C:\Users\PC\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp: In member function 'virtual int WiFiClient::connect(const char*, uint16_t)':
C:\Users\PC\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp:176:19: warning: comparison is always false due to limited range of data type [-Wtype-limits]
         if (_sock < 0) {
             ~~~~~~^~~
C:\Users\PC\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp: In member function 'virtual int WiFiClient::connect(IPAddress, uint16_t)':
C:\Users\PC\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp:192:15: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     if (_sock < 0) {
         ~~~~~~^~~
C:\Users\PC\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp: In member function 'virtual int WiFiClient::connectv6(IPv6Address, uint16_t)':
C:\Users\PC\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi\src\WiFiClient.cpp:206:15: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     if (_sock < 0) {
         ~~~~~~^~~

closed 2025-01-19 by tesa-klebeband