tesa-klebeband/RTL8720dn-Deauther #33: No Deauth Packets Detected on BW16-Kit USB-C

ap-apely, 2025-02-11:
### Steps to Reproduce:
1. Downloaded the code and set up Arduino IDE.  
2. Compiled and uploaded the code to my **BW16-Kit USB-C**.  
3. Compilation produced a few warnings, but I believe they do not affect functionality.  

### Issue:
- When launching an attack on my own network, no **deauth packets** are being detected, even on **2.4GHz** (checked via Wireshark).  

### Questions:
1. Are there any known fixes or troubleshooting steps for this issue?  
2. Does enabling **Log Rx and Log Tx** affect the functionality?  
3. In Arduino IDE, the board type is not detected automatically and must be selected manually—could this be related?  

### Additional Information:
- Connected via **USB-C** directly.  
- Using **BW16-Kit USB-C**.  

Any help or suggestions would be appreciated!

tesa-klebeband, 2025-02-11:
Thank you for bringing this to my attention and for actually structuring your issue!
For question 2 and 3 I can confidently say that they do not affect the expected functionality. Some people noticed in other issues that a specific purple version of the board does not work as there have apparently been some wrong PCB traces found, so that might be a problem. When you debugged it using wireshark, did you enter promiscuous mode on your WiFi card and did you make sure the channel you specified with airmon-ng (or any similar tool) is the same as the WiFis your trying to attack? And if yes, are you able to see other packets coming from the Mac address of the board?

Have a nice day

Janek79ax, 2025-02-17:
I recently did some careful tests with 2 boards: RTL8720DN BW16 (black) and RTL8720DN BW16E (purple). Scenario:
 - 2 networks (A and B) available with very similar MAC (first 7 hex letters are the same) on the same channel
 - only A is selected as a target
 - wireshark + airodump_ng monitoring the channel
 - what happened is that the other network B was attacked (devices disconnected from B only (not from A )and wire shark reveals correct packets with destination address B)
 - I’ve tested it multiple times on both devices - always the same single network B was attacked (even if I selected I was tied to attack different network C - still B was attacked)
 - this seems to be unrelated to reason code - eventually I was just leaving it empty

Janek79ax, 2025-02-23:
Ok, bit of good news @tesa-klebeband  - I was able to debug and partially fix Your code :)

What is wrong: current_num is always equal to zero. And it is used to determine which scan_results vector element is to be attacked - so it was always the very first row in scan results index. 

I’ve temporarily corrected it as:
While (deauth_wifis.size() > ) {
Current_num = deauth_wifis[0];
…

Since deauth_wifis[0] contains index of first selected checkbox, then it all goes well and memcpy takes the selected row and attacks it nicely :)

Downside is now I can attack only one network. At some stage next week I will try to modify Your code and attack multiple networks - one after another in a loop. 

——
Summary: complaints that it does not deauth are because it attacks the wrong network. 
——

Btw - great work - finally small 5GHz deauther - brilliant! Thanks! :)

tesa-klebeband, 2025-02-23:
Ohhh

ap-apely, 2025-02-24:
I confirm that this solved the problem. I apologize for not answering for a long time. Thank you all for your help!

tesa-klebeband, 2025-02-24:
Yes, I'll fix this ASAP. Thank you so much guys for helping out, I believe this was the issue with other people as well when they said that it didn't work.
Have a great day

Janek79ax, 2025-03-05:
Hey @tesa-klebeband , I've sent you a discord invite - please accept. I would like to discuss your another project - ESP32-deauther. I'm struggling to understand the idea of sniffer function comparing mac_header->dest, deauth_frame.sender and the idea of sending deauth frames only as a reaction to incoming traffic. 
Thanks
Janek