tesa-klebeband/RTL8720dn-Deauther #34: Flash the 2nd time the same equipment.

Neon288, 2025-02-19:
Goodnight.
After I uploaded it, everything works fine, now how do I do the procedure again on the same equipment, since the following sentence appears in the Arduino IDE:

Device Identifier is required for J-Link configurations. Please see https://www.segger.com/downloads/supported-devices.php for supported devices.

THX for your help.

Neon288, 2025-02-19:
[image]

tesa-klebeband, 2025-02-20:
I have no idea what is going on there 😅
It doesn't look like a problem with the actual code, probably just some driver issue on your side or problem with Arduino IDE. Maybe you can fix it by switching the version of Realtek Ameba Boards.

Neon288, 2025-02-23:
Hello. I changed the computer and everything goes well. I don't know why but um my laptop this message appears. 

Now with the firmware running, when i entered with the check in the checkbox and put the any reason and hit launch attack, nothing happens and the led blink green/red and not blue.
Any Idea? I tried to put another reason codes but nothing works.

THX for your help. Anything where i can help or test, feel free to contact!
Have a good week

tesa-klebeband, 2025-02-23:
Is your browser showing a "Page not found" or similar? If yes, try reloading the page and resubmit the form data. I've had this issue as well, but wasn't able to fix it, as it also seems to depend on the specific web browser. Hope this helps
Have a nice week too

closed 2025-03-11 by tesa-klebeband

Neon288, 2025-03-23:
Hello. I tested with several browsers, and the same thing happens in all of them. I access the page and select the network. I enter all the credentials, but the LED only blinks green/yellow. It never turns blue, and the network never disconnects.

Best regards