tesa-klebeband/RTL8720dn-Deauther #36: problem with uploading

maroixx, 2025-03-06:
In file included from C:\Users\MAROIX\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.8-build20250225\cores\ambd/Arduino.h:46:0,
                 from C:\Users\MAROIX\AppData\Local\arduino\sketches\EE00E7893F3640530239DCCE80790D02\sketch\RTL8720dn-Deauther.ino.cpp:1:
c:\users\maroix\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\bits\stl_vector.h: In member function 'std::vector<_Tp, _Alloc>::size_type std::vector<_Tp, _Alloc>::_M_check_len(std::vector<_Tp, _Alloc>::size_type, const char*) const':
C:\Users\MAROIX\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.8-build20250225\cores\ambd/wiring_constants.h:78:19: error: expected unqualified-id before '(' token
 #define max(a, b) ((a) > (b) ? (a) : (b))
                   ^
c:\users\maroix\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\bits\stl_bvector.h: In member function 'std::vector<bool, _Alloc>::size_type std::vector<bool, _Alloc>::_M_check_len(std::vector<bool, _Alloc>::size_type, const char*) const':
C:\Users\MAROIX\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.8-build20250225\cores\ambd/wiring_constants.h:78:19: error: expected unqualified-id before '(' token
 #define max(a, b) ((a) > (b) ? (a) : (b))
                   ^

exit status 1

Compilation error: exit status 1

tesa-klebeband, 2025-03-07:
Try adding an **#undef max** line before **#include "vector"**

maroixx, 2025-03-07:
> Try adding an **#undef max** line before **#include "vector"**

never mind bc i used 3.1.8 and not 3.1.7

maroixx, 2025-03-08:
> Ah, so its working now?

yeah

closed 2025-03-08 by maroixx