tesa-klebeband/RTL8720dn-Deauther #37: error not found

Pvtuan92, 2025-03-11:
I am a freshman in embedded programming, I tried to follow the instructions but the device is not working.

[image]

[image]
[image]
[image]

[image]

tesa-klebeband, 2025-03-11:
I know this issue exists but have no idea why. Reloading and choosing to resend form data seems to fix it. I'll have to investigate more on why this error even happens.

Pvtuan92, 2025-03-11:
Tks yes, thank you for your interest in my question

closed 2025-03-11 by Pvtuan92