tesa-klebeband/RTL8720dn-Deauther #42: Error

i11q, 2025-03-21:
I uploaded the code on the chip and the patch was successful, but the name of the private network does not appear, but a chip, why?

ycancook, 2025-03-22:
> I uploaded the code on the chip and the patch was successful, but the name of the private network does not appear, but a chip, why?

You have selected the wrong chip on the Arduino. Please choose the correct AI thinker chip and erase its memory first

i11q, 2025-03-22:
How do I delete the memory?

i11q, 2025-03-22:
> > I uploaded the code on the chip and the patch was successful, but the name of the private network does not appear, but a chip, why?
> 
> You have selected the wrong chip on the Arduino. Please choose the correct AI thinker chip and erase its memory first

How do I delete the memory?

ycancook, 2025-03-22:
> > > I uploaded the code on the chip and the patch was successful, but the name of the private network does not appear, but a chip, why?
> > 
> > 
> > You have selected the wrong chip on the Arduino. Please choose the correct AI thinker chip and erase its memory first
> 
> How do I delete the memory?

Ardruino - Tool - Erase All Flash ..... You see if your Board is BW16 or BW16E. If your board is BW16E, I would like to express my condolences because ardruino does not support running this code

i11q, 2025-03-22:
> Ardruino - Tool - Erase All Flash ..... You see if your Board is BW16 or BW16E. If your board is BW16E, I would like to express my condolences because ardruino does not support running this code

Yes it’s bw16 not bw16e

closed 2025-04-30 by tesa-klebeband