tesa-klebeband/RTL8720dn-Deauther #43: wifi_cust_tx

14mn0760d, 2025-03-22:
i'm always getting this error
fatal error: wifi_cust_tx.h: No such file or directory
 #include "wifi_cust_tx.h"
                          ^
compilation terminated.
exit status 1

Compilation error: wifi_cust_tx.h: No such file or directory

ycancook, 2025-03-22:
> wifi_cust_tx.h

See if the folder containing your .ino file has that file? If it doesn't exist, copy the file "wifi_cust_tx.h" into it

14mn0760d, 2025-03-22:
> > wifi_cust_tx.h
> 
> See if the folder containing your .ino file has that file? If it doesn't exist, copy the file "wifi_cust_tx.h" into it

the file is there, i redownload the file and compile and now i get this error: 

C:\Users\Nelson\AppData\Local\arduino\sketches\774DD9B8EC494498CD28755CCB4D3397\libraries\WiFi\wifi_cust_tx.cpp.o: In function `wifi_tx_raw_frame(void*, unsigned int)':
wifi_cust_tx.cpp:(.text._Z17wifi_tx_raw_framePvj+0x0): multiple definition of `wifi_tx_raw_frame(void*, unsigned int)'
C:\Users\Nelson\AppData\Local\arduino\sketches\774DD9B8EC494498CD28755CCB4D3397\sketch\wifi_cust_tx.cpp.o:wifi_cust_tx.cpp:(.text._Z17wifi_tx_raw_framePvj+0x0): first defined here
C:\Users\Nelson\AppData\Local\arduino\sketches\774DD9B8EC494498CD28755CCB4D3397\libraries\WiFi\wifi_cust_tx.cpp.o: In function `wifi_tx_deauth_frame(void*, void*, unsigned short)':
wifi_cust_tx.cpp:(.text._Z20wifi_tx_deauth_framePvS_t+0x0): multiple definition of `wifi_tx_deauth_frame(void*, void*, unsigned short)'
C:\Users\Nelson\AppData\Local\arduino\sketches\774DD9B8EC494498CD28755CCB4D3397\sketch\wifi_cust_tx.cpp.o:wifi_cust_tx.cpp:(.text._Z20wifi_tx_deauth_framePvS_t+0x0): first defined here
C:\Users\Nelson\AppData\Local\arduino\sketches\774DD9B8EC494498CD28755CCB4D3397\libraries\WiFi\wifi_cust_tx.cpp.o: In function `wifi_tx_beacon_frame(void*, void*, char const*)':
wifi_cust_tx.cpp:(.text._Z20wifi_tx_beacon_framePvS_PKc+0x0): multiple definition of `wifi_tx_beacon_frame(void*, void*, char const*)'
C:\Users\Nelson\AppData\Local\arduino\sketches\774DD9B8EC494498CD28755CCB4D3397\sketch\wifi_cust_tx.cpp.o:wifi_cust_tx.cpp:(.text._Z20wifi_tx_beacon_framePvS_PKc+0x0): first defined here
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

ycancook, 2025-03-22:
> the file is there, i redownload the file and compile and now i get this error:
>
> C:\Users\Nelson\AppData\Local\arduino\sketches\774DD9B8EC494498CD28755CCB4D3397\libraries\WiFi\wifi_cust_tx.cpp.o: In [...]

try deleting the folder \Users\Nelson\AppData\Local\arduino\sketches\774DD9B8EC494498CD28755CCB4D3397

14mn0760d, 2025-03-22:
> try deleting the folder \Users\Nelson\AppData\Local\arduino\sketches\774DD9B8EC494498CD28755CCB4D3397

i deleted still get the same error i think, i appreciate you helping me thanks a lot

C:\Users\Nelson\AppData\Local\arduino\sketches\D8B7033E51EC6B2A0EF5DD648E0CF8A0\libraries\WiFi\wifi_cust_tx.cpp.o: In function `wifi_tx_raw_frame(void*, unsigned int)':
wifi_cust_tx.cpp:(.text._Z17wifi_tx_raw_framePvj+0x0): multiple definition of `wifi_tx_raw_frame(void*, unsigned int)'
C:\Users\Nelson\AppData\Local\arduino\sketches\D8B7033E51EC6B2A0EF5DD648E0CF8A0\sketch\wifi_cust_tx.cpp.o:wifi_cust_tx.cpp:(.text._Z17wifi_tx_raw_framePvj+0x0): first defined here
C:\Users\Nelson\AppData\Local\arduino\sketches\D8B7033E51EC6B2A0EF5DD648E0CF8A0\libraries\WiFi\wifi_cust_tx.cpp.o: In function `wifi_tx_deauth_frame(void*, void*, unsigned short)':
wifi_cust_tx.cpp:(.text._Z20wifi_tx_deauth_framePvS_t+0x0): multiple definition of `wifi_tx_deauth_frame(void*, void*, unsigned short)'
C:\Users\Nelson\AppData\Local\arduino\sketches\D8B7033E51EC6B2A0EF5DD648E0CF8A0\sketch\wifi_cust_tx.cpp.o:wifi_cust_tx.cpp:(.text._Z20wifi_tx_deauth_framePvS_t+0x0): first defined here
C:\Users\Nelson\AppData\Local\arduino\sketches\D8B7033E51EC6B2A0EF5DD648E0CF8A0\libraries\WiFi\wifi_cust_tx.cpp.o: In function `wifi_tx_beacon_frame(void*, void*, char const*)':
wifi_cust_tx.cpp:(.text._Z20wifi_tx_beacon_framePvS_PKc+0x0): multiple definition of `wifi_tx_beacon_frame(void*, void*, char const*)'
C:\Users\Nelson\AppData\Local\arduino\sketches\D8B7033E51EC6B2A0EF5DD648E0CF8A0\sketch\wifi_cust_tx.cpp.o:wifi_cust_tx.cpp:(.text._Z20wifi_tx_beacon_framePvS_PKc+0x0): first defined here
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

14mn0760d, 2025-03-22:
> try deleting the folder \Users\Nelson\AppData\Local\arduino\sketches\774DD9B8EC494498CD28755CCB4D3397

i deleted and still get these 

C:\Users\Nelson\AppData\Local\arduino\sketches\D8B7033E51EC6B2A0EF5DD648E0CF8A0\libraries\WiFi\wifi_cust_tx.cpp.o: In function `wifi_tx_raw_frame(void*, unsigned int)':
wifi_cust_tx.cpp:(.text._Z17wifi_tx_raw_framePvj+0x0): multiple definition of `wifi_tx_raw_frame(void*, unsigned int)'
C:\Users\Nelson\AppData\Local\arduino\sketches\D8B7033E51EC6B2A0EF5DD648E0CF8A0\sketch\wifi_cust_tx.cpp.o:wifi_cust_tx.cpp:(.text._Z17wifi_tx_raw_framePvj+0x0): first defined here
C:\Users\Nelson\AppData\Local\arduino\sketches\D8B7033E51EC6B2A0EF5DD648E0CF8A0\libraries\WiFi\wifi_cust_tx.cpp.o: In function `wifi_tx_deauth_frame(void*, void*, unsigned short)':
wifi_cust_tx.cpp:(.text._Z20wifi_tx_deauth_framePvS_t+0x0): multiple definition of `wifi_tx_deauth_frame(void*, void*, unsigned short)'
C:\Users\Nelson\AppData\Local\arduino\sketches\D8B7033E51EC6B2A0EF5DD648E0CF8A0\sketch\wifi_cust_tx.cpp.o:wifi_cust_tx.cpp:(.text._Z20wifi_tx_deauth_framePvS_t+0x0): first defined here
C:\Users\Nelson\AppData\Local\arduino\sketches\D8B7033E51EC6B2A0EF5DD648E0CF8A0\libraries\WiFi\wifi_cust_tx.cpp.o: In function `wifi_tx_beacon_frame(void*, void*, char const*)':
wifi_cust_tx.cpp:(.text._Z20wifi_tx_beacon_framePvS_PKc+0x0): multiple definition of `wifi_tx_beacon_frame(void*, void*, char const*)'
C:\Users\Nelson\AppData\Local\arduino\sketches\D8B7033E51EC6B2A0EF5DD648E0CF8A0\sketch\wifi_cust_tx.cpp.o:wifi_cust_tx.cpp:(.text._Z20wifi_tx_beacon_framePvS_PKc+0x0): first defined here
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

closed 2025-03-22 by 14mn0760d