tesa-klebeband/RTL8720dn-Deauther #49: [ERROR] Accept connection failed solution

viniciussilverio, 2025-04-01:
Hi,
First of all, thanks for the amazing work.
Now, regarding the error mentioned on the title, I found a solution, it requires changing the library, but it works. As I'm implementing the control over serial for my implementation I decided to share it in here, in case it helps someone else. See solution below:
https://forum.amebaiot.com/t/amb82-mini-arduino-monitor-filled-with-unwanted-debug-messages/3354/4

The path changes a little bit, but it is the same file, just need to replace the version in the path and you will find.

Again, thanks for the great work.

tesa-klebeband, 2025-04-01:
Thanks for discovering this, however it's a bit hard for me to fix it as I am not the owner of the code and don't provide the files. I could, however, provide a patch to comment the line out. But thanks for investigating 🤝

viniciussilverio, 2025-04-01:
Yes, totally agree, I will see if I can figure out a way to fix it without changing the library, if I have any lucky I will open a pull request. Thx again

tesa-klebeband, 2025-04-01:
No problem, thank you as well. Hope you'll succeed

closed 2025-04-01 by tesa-klebeband