tesa-klebeband/RTL8720dn-Deauther #50: deauthing is not working

Driv28, 2025-04-03:
Unfortunately, deauthing is not working. I selected three devices that are all connected to the same router (2.4GHz and 5GHz). I can still receive data on these devices.

tesa-klebeband, 2025-04-03:
Hmm that's unfortunate. Could you tell me which board you are using? And have you modified the software in any way?

Driv28, 2025-04-03:
i use the Ai-Thinker BW16 WiFi. I didn't modified the software.

sitirka1, 2025-04-04:
Most RTL8720dn devboards have the upload and log serial ports isolated. Make sure to bridge LOG_UART_TX and LP_UART_TX, as well as LOG_UART_RX and LP_UART_RX during uploading. how ????

Driv28, 2025-04-04:
> Most RTL8720dn devboards have the upload and log serial ports isolated. Make sure to bridge LOG_UART_TX and LP_UART_TX, as well as LOG_UART_RX and LP_UART_RX during uploading. how ????

İf you also have the AI Thinkerboard. Just press and hold the burn/flash button, than press the reset button once and also release the burn/Flash Button.

sitirka1, 2025-04-04:
deauthing is not working why? i selected 1 or 2 SSID, i am still connected

tesa-klebeband, 2025-04-04:
What reason code did you use?

sitirka1, 2025-04-04:
> What reason code did you use?
i tried 2,1,0

tesa-klebeband, 2025-04-04:
Try 6 and 23, maybe they work.

sitirka1, 2025-04-05:
> Try 6 and 23, maybe they work.

still the same

tesa-klebeband, 2025-04-05:
Are you sure your WiFi isn't using wpa3? Wpa3 networks cannot be deauthed using this project. Have you tried using a different browser and checked for solutions in the closed issues?

Justers1, 2025-04-23:
Same issue. Tried on wpa2 no any results - showing green light but network is working. 
Here is some logs:
Starting AP ...
Realm started
 [INFO] server_drv.cpp: WiFi server is set to non-blocking mode
 [INFO] Listen socket successfully
[INFO] Socket connect successfully 
 [ERROR] Accept connection failed
 [ERROR] Accept connection failed
 [ERROR] Accept connection failed

tesa-klebeband, 2025-04-23:
Do you see the light flashing on your device after the deauth attack has been started?

Justers1, 2025-04-23:
It blinks purple a couple of times, then stays solid green.
[image]

tesa-klebeband, 2025-04-30:
Looks like an issue with the webserver. Maybe try a different browser. When deauthing it should blink blue. You can also switch to an older commit and see if that fixes it, there was a rewrite on the webserver logic, so there may be unknown bugs.

Justers1, 2025-04-30:
> Looks like an issue with the webserver. Maybe try a different browser. When deauthing it should blink blue. You can also switch to an older commit and see if that fixes it, there was a rewrite on the webserver logic, so there may be unknown bugs.

Thanks, but I’ve tried almost all versions and the same issue persists everywhere. The blue light turns on for only 2 seconds, and then the green light stays on.

[ERROR] Accept connection failed
 [INFO] Accept connection successfully
 A client connected to this server :
[PORT]: 9066
[IP]:192.168.1.101
 [INFO] Accept connection successfully
 A client connected to this server :
[PORT]: 9067
[IP]:192.168.1.101
 [ERROR] Accept connection failed
 [ERROR] Accept connection failed
 [ERROR] Accept connection failed
 [ERROR] Accept connection failed

ycancook, 2025-05-01:
I tested on a number of wireless routers and found that the ASUS AX1800 and higher routers have very high resistance to deauther. I wonder which router you guys are using for your science experiments

Justers1, 2025-05-03:
> I tested on a number of wireless routers and found that the ASUS AX1800 and higher routers have very high resistance to deauther. I wonder which router you guys are using for your science experiments

TP-Link. I also tried all my other routers, tested access points from all my smartphones, and even found an old Huawei E5573 GSM modem. Same result.

ycancook, 2025-05-03:
> TP-Link. I also tried all my other routers, tested access points from all my smartphones, and even found an old Huawei E5573 GSM modem. Same result.

You check whether the network you are attacking is a single network or a dual network. If the network is dual, the attack mechanism must not delay to push it out of the device. And try editing the code int current_channel = 1; to int current_channel = 40; try it (this helps focus on the 5Ghz frequency when attacking)

Justers1, 2025-05-04:
> You check whether the network you are attacking is a single network or a dual network. If the network is dual, the attack mechanism must not delay to push it out of the device. And try editing the [...]

Maybe there's something wrong with my board, because whatever I try and whichever network I use, the result is always the same.

[image]
[image]

tesa-klebeband, 2025-05-04:
> You check whether the network you are attacking is a single network or a dual network. If the network is dual, the attack mechanism must not delay to push it out of the device. And try editing the [...]

Changing current channel doesn't affect the logic that handles the deauthing as the channel is adjusted to the one of the network automatically.

tesa-klebeband, 2025-05-04:
> Maybe there's something wrong with my board, because whatever I try and whichever network I use, the result is always the same.
>
> [image]
> [image]

Your board looks fine I think. The problem is that the blue light doesn't stay on permanently. Are there any logs after the blue led turns off? If you are able to, you can try sending the requests to the Webserver manually and see if that fixes it. If yes, then it would be an issue with your browser.

ycancook, 2025-05-04:
> Changing current channel doesn't affect the logic that handles the deauthing as the channel is adjusted to the one of the network automatically.

That's the theory, but for some reason, in practice, I tried it and found that setting current_channel > or = 36 makes the attack more effective. I'll have a video to prove it.

ycancook, 2025-05-04:
> Changing current channel doesn't affect the logic that handles the deauthing as the channel is adjusted to the one of the network automatically.

I have tried many times like this video of mine
https://www.youtube.com/watch?v=t7Pc-c7e3pQ

Justers1, 2025-05-04:
> Your board looks fine I think. The problem is that the blue light doesn't stay on permanently. Are there any logs after the blue led turns off? If you are able to, you can try sending the requests [...]

Unfortunately, I'm unable to send a request to the web server manually.
When I flash the board, some errors appear — maybe that's why everything isn't working as it should.

[image]

Only these logs:
[ERROR] Accept connection failed
[INFO] Accept connection successfully
A client connected to this server :
[PORT]: 8253
[IP]:192.168.1.100
[ERROR] Accept connection failed
[INFO] Accept connection successfully
A client connected to this server :
[PORT]: 8256
[IP]:192.168.1.100
[ERROR] Accept connection failed
[ERROR] Accept connection failed
[ERROR] Accept connection failed

ycancook, 2025-05-04:
> Unfortunately, I'm unable to send a request to the web server manually. When I flash the board, some errors appear — maybe that's why everything isn't working as it should.
>
> [image]
>
> Only these [...]

That's not an error, that's a warning about variables in the WiFiClient.cpp library (you can leave it alone and it runs fine). When running the [ERROR] Accept connection failed log many times, it is very normal, even the sample code in the adruino library is the same.  

I think when you have finished loading the code, turn off the Arduino and use the Firefox browser to test it, I find this browser the most stable to test the code.

Justers1, 2025-05-04:
> I think when you have finished loading the code, turn off the Arduino and use the Firefox browser to test it, I find this browser the most stable to test the code.

I’ve tried that too… ugh, I give up. Thanks for the advice anyway.

closed 2025-05-21 by tesa-klebeband