tesa-klebeband/RTL8720dn-Deauther #54: Only 2.4 Ghz avalible?

Matrix-84, 2025-04-23:
I flashed the firmware on the BW16 (RTL8720DN) as described, but unfortunately, only 2.4 GHz networks are displayed, but no 5 GHz network.
The BW16 (RTL8720DN) is located right next to the 5 GHz router, but only the 2.4 GHz network is displayed.

Does anyone know a workaround for this problem?

ycancook, 2025-04-28:
> I flashed the firmware on the BW16**E** (RTL8720DN) as described, but unfortunately, only 2.4 GHz networks are displayed, but no 5 GHz network. The BW16**E** (RTL8720DN) is located right next to the 5 GHz router, but only the 2.4 GHz network is displayed.
> 
> Does anyone know a workaround for this problem?
You should keep the BW16E cabinet as a collection, and buy the BW16 to use

tesa-klebeband, 2025-04-30:
Yes, BW16E is not supported

closed 2025-04-30 by tesa-klebeband